ziutek/emgo #5: Generating packages for stm32f030x8

ironsteel, 2018-03-31:
Hello,

first of, really great and interesting project. Thanks for the awesome work  

I'm trying to add support for **stm32f030x8** but I can't find any documentation on how to do it. I found a script in `emgo/egpath/src/stm32/xgen/run.sh` and added 

```sh
stm32xgen stm32/o/f030x8 <stm32f0xx/stm32f030x8.h
``` 
line to it.

I ran the script, it generated some errors but I didn't find for example the `f030x8--mmap.go` file in:

```sh
emgo/egpath/src/stm32/hal/raw/mmap
```

I've successfully installed `xgen` and `stm32xgen` in my $GOPATH, so they are running fine and are found by the script.

The errors which generated can be found in this gist: https://gist.github.com/ironsteel/64bdaa091b2593e7ce6225e25faa4170

Help would be really appreciated. If you need more info I'll be happy to provide it to you.

ziutek, 2018-03-31:
stm32xgen doesn't handle all quirks in ST HAL, hence errors. You need to cd to stm32/hal and run ./mkraw.sh to rebuild stm32/hal/raw.

There is no full HAL support for F0 now. Review *f0*.go files in stm32/hal to see what is supported. Pay attention to +build tags. You can always use automatically generated packages in stm32/hal/raw. 

Try to determine differences between f030x6 and f030x8 and write your findings there. This can help to add support to it. You can do more and study differences to F303. It seems that whole code for F3 can be used for F0, only peripheral instantiation is different.

To add support for new target you need to add new tag to // +build or create new file with this tag (especially for peripheral instances).

The linker script for f030x8 is ready. If you have other MCUs, add corresponding files to egpath/ld/stm32. We use naming from ST header files (all lowercase).

I've just generated raw packages for f030x8. Just pull from devel branch and work on this.

Active development takes place in devel branch.

ironsteel, 2018-04-02:
Thank you very much for the instructions, hints and generating the needed files.

> To add support for new target you need to add new tag to // +build or create new file with this tag (especially for peripheral instances).
Can you point me to an existing file?

Thank you!

ziutek, 2018-04-04:
Try to study files in *egpath/stm32/hal*.

For example there is file *egpath/src/stm32/hal/tim/instances-f0.go*

It contains instances of TIM peripheral and currently handles both f030x6 and f030x8. But f030x8 has more timers than f030x4 and f030x6 (7 instead of 5). We can split this file to two: instances-f030x6.go, instances-f030x8.go but I do not like this solution very much.

Maybe support f030x8 target only (or higher) as long as its set of peripherals is a superset of smaller ones. Even ST doesn't treat separately f030x4 - it is handled together with f030x6, but it seems that they differ only in the number of ADC channels.

http://www.st.com/en/microcontrollers/stm32f0x0-value-line.html?querycriteria=productId=LN1826

ipfans, 2018-04-16:
I still no idea to support new platform. E.g. I want to add support for STM32F107, so which file should I modify?

ziutek, 2018-04-16:
There is no single file to edit.

Minimal set of things to do:

1. Add a new target to `emgo/egpath/src/stm32/xgen/run.sh` and run it. It generates packages in `emgo/egpath/src/stm32/o/yuor_new_mcu`.

2. Create the linker script for new MCU in `emgo/egpath/ld/stm32`.

Done.

Now you can import raw packages from `stm32/o/yuor_new_mcu`. You can't use STM32 HAL so yo u don't need to set EGTARGET.

To support new MCU in `stm32/hal`:

3. Run `egpath/src/stm32/hal/mkraw.sh` to generate packages in `egpath/src/stm32/hal/raw`.

4. Review and edit packages in `egpath/src/stm32/hal`.

You probably should start from `stm32/hal/system`. There is a good chance that your new MCU has the same or similar RCC peripheral to other already supported MCUs. In this case you need only add new target to the `// +build` directive in eg. `setup-f1.go` and `clk-f1.go`. If it differs, you have to write new `SetupPLL` function for it in new `setup-xxx.go` file and set `HSIClk`, `maxAPB1Clk`, `maxAPB2Clk` in `clk-xxx.go`.

Set EGTARGET environment variable to new board and try compile minimal program that only calls `system.SetupPLL`  and `systick.Setup`. If you succeeded review end edit `stm32/hal/gpio` to add support for GPIO and so on...