ziutek/emgo #7: Does nrf5 ble support hid?

dzhou121, 2018-04-16:


ziutek, 2018-04-16:
There i n=

ziutek, 2018-04-16:
There is no ready to use BLE package in Emgo.

dzhou121, 2018-04-17:
Is it easy to add? And I can help if you could give me a bit of guidance.

ziutek, 2018-04-17:
No, it isn't easy... at least for me.

The problem isn't higher layers but lowest layer: BLE controller.

You need to read:

1. Bluetooth Core Specification (at least 4.0): https://www.bluetooth.com/specifications/bluetooth-core-specification/archived-specifications

2. nRF51822 (preferably also nRF52832) Reference Manual: http://infocenter.nordicsemi.com/pdf/nRF51_RM_v3.0.1.pdf

The problem insist in **strict timing**. You have to design a state machine that properly handles **all cases** and you need to write it cleanly in Go. You need a deep understanding of the protocol and hardware. Debuging is difficult, especially if your only tool is Android phone with its HCI logging.

There are open source implementations in: [MyNewt](https://mynewt.apache.org/) and [Zephyr](https://www.zephyrproject.org/). The last one has some support from Nordic: https://devzone.nordicsemi.com/b/blog/posts/nrf5x-support-within-the-zephyr-project-rtos

I didn't studied their code but it's probably worth doing.

There is my proof of concept code in [nrf5/blec](https://github.com/ziutek/emgo/tree/master/egpath/src/nrf5/blec) and an [example](https://github.com/ziutek/emgo/tree/master/egpath/src/nrf5/examples/core51822/ble-connect) that almost works. But everything must be be thought over and rewritten.