ziutek/emgo #8: [Question] Steps to capture SWO

sergiuszm, 2018-04-24:
Hello Michal,

In README.MD you wrote: 
> There is a set of scripts for any board in example directory that simplifies loding and debuging process. The load-oocd.sh and swo-oocd.sh scripts can handle SWO output from ITM (Instrumentation Trace Macrocell) but needs itmsplit to convert binary stream to readable messages. SWO is very useful for debuging and fmt.Print* functions by default use ITM trace port as standard output.

Could you write in steps how to get that output? I'm using usart example with Nucleo-L476RG but after some time my board stops working so I'm trying to debug it. Unfortunately, I haven't figured out how to get SWO working - I have tried your scripts but it seems that I am not able to get it working.

I would really appreciate if you could write steps starting from flashing a microcontroller (maybe I am just missing something).

ziutek, 2018-04-24:
I don't want to write there a general explanation. This is a topic for a separate article and I probably write about it on my blog.

This is issue tracker, so please write:

1. What exactly you are doing.
2. What you obtain.
3. What you expect.

I will try to help.

sergiuszm, 2018-04-25:
I have tried to debug a case in which a serial communication stopped working on the emgo side. In that case debugging what is happening on microcontroller would be nice but I was not able to use prepared by you scripts: debug-oocd, debug-stutil or swo-oocd so I gave up. It would be nice to read about it on your blog so I am looking forward to it.

Fortunately, I figured out that there is something wrong with my laptop (macbook) itself that causes that issue which doesn't exist while I am using a PC with linux.

I am planning to add emgo support for MSP430FR5994 in the future so I will be looking for your guidance for sure (if you don't mind) :)

closed 2018-04-25 by sergiuszm