ziutek/emgo #9: Goroutines: undefined reference to `internal$NewTask'

sergiuszm, 2018-04-25:
Michal, I am trying to compile a code with go routine - a very similar to the one which you have presented on your blog.

```
package main

import (
	"delay"
	"fmt"

	"stm32/hal/gpio"
	"stm32/hal/system"
	"stm32/hal/system/timer/systick"
)

var led, led2 gpio.Pin

func init() {
	system.SetupPLL(-48, 6, 20, 0, 0, 2) // 80 MHz (max. for voltage Range 1).
	//system.Setup80(0, 0) // Short form of above configuration.
	//system.SetupPLL(-4, 1, 26, 0, 0, 4) // 26 MHz (max. for voltage Range 2).
	system.SetupMSI(100) // Lowest possible frequency (100 kHz).
	systick.Setup(2e7) // Typical 2e6 ns is to low for 100 kHz SysClk.

	gpio.A.EnableClock(false)
	led = gpio.A.Pin(5)
	led2 = gpio.A.Pin(6)

	cfg := gpio.Config{Mode: gpio.Out, Speed: gpio.Low}
	led.Setup(&cfg)
	// cfg2 := gpio.Config{Mode: gpio.Out, Driver: gpio.OpenDrain}
	led2.Setup(&cfg)
}

func blinky(led gpio.Pin, period int) {
	for {
		led.Clear()
		delay.Millisec(100)
		led.Set()
		delay.Millisec(period - 100)
	}
}

func main() {
	delay.Millisec(500)
	buses := []system.Bus{
		system.Core,
		system.AHB,
		system.APB1,
		system.APB2,
	}
	fmt.Printf("\r\n")
	for _, bus := range buses {
		fmt.Printf("%4s: %9d Hz\r\n", bus, bus.Clock())
	}
	go blinky(led, 500)
	blinky(led2, 1000)
}
```
Unfortunately, I am getting compilation errors all the time (even for the example code from: "f030-demo-board/demo").

```
/tmp/eg-build977705888/main/__noos_cortexm4f_l476xx.o: In function `main$main':
emgo/egpath/src/stm32/examples/nucleo-l476rg/blinky/__noos_cortexm4f_l476xx.c:64: undefined reference to `internal$NewTask'
```

```
/tmp/eg-build490269123/main/__noos_cortexm0_f030x6.o: In function `main$main':
emgo/egpath/src/stm32/examples/f030-demo-board/demo/__noos_cortexm0_f030x6.c:88: undefined reference to `internal$NewTask'
emgo/egpath/src/stm32/examples/f030-demo-board/demo/__noos_cortexm0_f030x6.c:96: undefined reference to `internal$NewTask'
```

I am using the newest gcc-arm-none-eabi toolchain: gcc-arm-none-eabi-7-2017-q4-major and go version 1.9.5.

Am I doing something wrong or is there something missing in the repository?

ziutek, 2018-04-25:
Hmm. I need to review this.

I tried compile your code on my system:

```
$ EGOS=noos EGARCH=cortexm4f EGTARGET=l476xx egc
$ arm-none-eabi-size cortexm4f.elf 
   text    data     bss     dec     hex filename
  31812     324     348   32484    7ee4 cortexm4f.elf
```

This is real output for your *main.go* and *script.ld*.

Ensure that EGCC, EGLD, EGAR points to arm-none-eabi binaries and EGROOT, EGPATH are set correctly.

Regarding SWO, you have a bug in your code. You call *system.Setup* two times. I don't know what this causes but LED blinks. Comment the line:

```go
system.SetupMSI(100)
```

and compile the program again. After that, load it with this magic command (you need itmsplit on your path):

```
$ openocd -d0 -f interface/stlink.cfg -f target/stm32l4x.cfg -c 'init; reset init; program cortexm4f.elf; tpiu config internal /dev/stdout uart off 80000000; itm ports on; reset run' |itmsplit /dev/stdout
```
This is what *load-oocd.sh* script does for this specific case (board, speed, etc). And there is its output:

```
Open On-Chip Debugger 0.10.0+dev-00319-g8f1f912a (2018-03-07-19:20)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080051b0 msp: 0x20003480
adapter speed: 4000 kHz
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080051b0 msp: 0x20003480
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000050 msp: 0x20003480
wrote 32768 bytes from file cortexm4f.elf in 1.551383s (20.627 KiB/s)
** Programming Finished **
adapter speed: 480 kHz

Core:  80000000 Hz
 AHB:  80000000 Hz
APB1:  80000000 Hz
APB2:  80000000 Hz
```

ziutek, 2018-04-25:
You're right with *always_inline*. Virtually any *inline* directive in any *+.h file must have this attribute or be *static inline*. I've using find and grep to find such bugs and found three. All fixed in last commit.

closed 2018-05-22 by ziutek