ziutek/emgo #10: how could I port it to arm a53

bugparty, 2018-05-21:


ziutek, 2018-05-22:
You have to port to whole SOC not only to Cortex A53.

If you really want, you should start from study the memory organisation of your SOC to write proper linker script. You also need to study its peripherals and booting process.

After that you probably will not have any problems with Emgo. Start from egroot/internal to implement basic functions like memory copying. After that go to runtime/noos and implement all code in *-cortexm* files.

The Cortex A53 based boards can for sure run full Go language. See: 

https://github.com/ycoroneos/golang_embedded
https://github.com/ycoroneos/G.E.R.T

bugparty, 2018-05-22:
thank you sir, a53 can run full go, but most of the routers just have a 8m flash,so I found your project

closed 2018-05-22 by bugparty