ziutek/emgo #15: how to use xgen

wumingyu12, 2018-06-25:
hello，first it is a wonderful project，i has hate to use c to write Embedded program。

i find you use xgen to create sdk .go file by register config .go file。
now i want to add something in register config .go file 
how can i use xgen to update sdk .go file like xgen_ficr.go

ziutek, 2018-06-25:
Simly run:

```
xgen ficr.go
```

It regenerates **xgen_ficr.go**.

For STM32 see issue #13

ziutek, 2018-06-25:
There is one unobvious thing.

Bit field names must have a comment beginning with "//+". See the following example:

```go
const (
    SPEED    = 3 << 5 //+
    VERYSLOW = 0 << 5
    SLOW     = 1 << 5
    NORMAL   = 2 << 5
    FAST     = 3 << 5
    DIR      = 1 << 7 //+
)
```
xgen uses it to distinguish between field name and its values. Of course, it is good to put a full description of any bit/bitfield.

wumingyu12, 2018-06-25:
i see and Thank you for your reply。