ziutek/emgo #16: how to use

wumingyu12, 2018-06-25:
in nrf5 example i find two example ble-advert and ble-connect，i cant understand how to work.

in my mind,i think it  use bluetooth protocol stack in nrf5 flash. but it seem it use your own bluetooth protocol stack.

my problem is how can i use  bluetooth protocol stack in nrf5 flash like nrf5 sdk.

if it is not a good way how can i use bluetooth.  for example set  Advertising interval ，Advertising_Type。

ziutek, 2018-06-25:
Sorry. There is no ready to use BLE stack.

ble-advert dosn't require any stack. It simply advertises (use nRF Connect to see something) and can be used as example how to write beacon application.

ble-connect uses Emgo BLE stack (not finished yet). You can connect to it using nRF Connect. Every attribute read returns "Hellow world" string.

Emgo isn't compatible with Nordic soft device. You have to mass erase Flash before load any example code.

See also issue #7.

ziutek, 2018-06-26:
See the beginning of my first comment in this topic:

**Sorry. There is no ready to use BLE stack.**

BLE stack (as this whole project) is under construction. There are things more stable than others in Emgo but BLE stack isn't even finished yet. No more than 25% functionality is implemented, only things that are need by ble-connect example.

The nrf5/blec package need to be rewritten because it doesn't fit well with the nRF hardware (RADIO, TIMERS, PPI). Currently it doesn't properly handle missing packets and doesn't support encryption.

ziutek, 2018-06-26:
The code in emgo/src/bluetooth contains hardware independent layers of Bluetooth stack. You can try extend and modify it. The ble.HCI  defines interface to communication with hardware:

```go
// HCI defines Host Controller Interface for Bluetooth LE.
type HCI interface {
	// Recv returns next received PDU.
	Recv() (DataPDU, error)

	// GetSend returns current transmit PDU. This PDU is not zeroed (can
	// contain random data in payload and header).
	GetSend() DataPDU

	// Send sends current transmit PDU. 
	Send() error
}
```
The nrf5/blec tries to implement this interface for nRF5 hardware. I think that the better and more reliable code is need there before work with higher layers but even now you can improve and extend  higher layers but you have to be prepared for problems from blec.