ziutek/emgo #18: wasm targets

mattkanwisher, 2018-11-12:
Since the main Go doesn't compile to native wasm, only a variant that runs in a web browser. I was wondering if this project had any plans of supporting native wasm target

ziutek, 2018-11-13:
Emgo is based on Go to C translation, and uses gcc as its backend. So if gcc can handle wasm target it is probably possible.

But this project is targeted to low-end 32-bit MCU and there are no any plans to support web browsers.

gedw99, 2018-12-01:
hey all. I was going to ask the same thing.

as you might knwo there is TinyGo thats designed also for golang embedded and it also is pushing to get WASM working too :)

https://github.com/aykevl/tinygo

lots of stuff about WSM there. For example: https://github.com/aykevl/tinygo/issues/82

emgo looks really cool too. Why not compile strengths ?

ziutek, 2018-12-05:
See #19